rstephan/ArtnetnodeWifi #1: get channel value

gamecompiler, 2016-09-09:
Hi rstephan,
did you implemend (my english is bad - i mean did you "already") a function to get the value of a channel ?
Your LED sketch isnt working because the leds will never be set....
Thats why im asking.

Thanks in advance,
gamecompiler

rstephan, 2016-09-09:
Hi gamecompiler,
i have not written the example "ESP8266_Artnet_RGB_LED".
https://github.com/solexious/ESP8266_artnet_led_node
You are right, the example is not working out-of-the-box. Lines 124 to 126 are the problems. But, if you un-comment the lines it wouldn't work either. Thanks for the hint, i will fix it later the day.
rstephan

gamecompiler, 2016-09-17:
Thank you very much, i will test it soon.
But what Channel is Frame 0 ?
As i know Channel 0 is not a Channel in DMX512 protocol.
(Sorry for my bad english)

rstephan, 2016-09-17:
DMX has a Start-Code, byte 0, with a value of 0, most of the time. Byte 1 is channel 1 and so on.
Take a look at my ArtnetnodeWifiSendDmx example.
Art-net don't know anything about the Start-Code. The art-net frame has only channel data. So byte 0 is channel 1 and so on.

gamecompiler, 2016-11-12:
Thank you very much! It works.

But maybe im just tired, or the latency is better if adding yield(); to the end of the code.
Also disabling ticker.attach(5, beat); made some improvements.
Overclocking the esp8266 to 180mhz made a huge improvement.
(There is something wrong with the Arduino esp8266 pwm, its not good Pwm, take a look at it with a oscilloscope, just the same shit using arduino IDE on a ATmega. You need to know im a little bit "Lightphobic")

What does the function artnetnode.setDMXOutput(0,1,0); do ? Can i enable direct Dmx output?
I thought about adding DMX functionality using the SimpleDMX library.

rstephan, 2016-11-14:
I have simplified the example, based on your comment.
The "*Output" methods are parts of the DMX output capabilities from the original author. The library was originally designed for an Arduino mega or so with more UARTs to send DMX bundled within the art net-lib. I have disabled all the stuff for the mega so the ESP can do it. It is better to separate these things, so i wrote my own DMX example (ArtnetnodeWifiSendDmx) to show how you can write your own "driver".
Feel free to derive my class or use the callback way to put DMX on the cable.

gamecompiler, 2016-11-14:
Ok thank you very much. Im currently writing a Artnet Node with DMX capabilitiys. (Using ESPDMX library).
Will test it tommorow. If it works, i will made a Pull Request.

Is there any way to check "which"**/**"if" channels have changed ? Im currently iterating through 0 to 511 and writing the Artnet values to ESPDMXs buffer. Not very efficent. Im searching for sth that (as example) returns me a matrix with only what changed.

Thanks in advance.

closed 2018-01-20 by gamecompiler