rstephan/ArtnetnodeWifi #2: LED flickering

haaslukas, 2017-04-04:
I've just downloaded the ArtnetnodeWifi library and tested the ESP8266_Artnet_RGB_LED.ino example on two different WeMos chips (based on ESP-8266ex). The node connects to my WLAN and receives the ArtNet packets. But all LEDs are flickering - regardless of the DMX value (except 0 of course, where all LEDs are turned off).

Looks like PWM problem to me, but can't figure out how to solve it for the moment. Already tried to set the PWM frequency to 200Hz by adding analogWriteFreq(200); but it didn't help. Does anyone have an idea? Is it an ESP8266 problem or maybe an library issue?

rstephan, 2017-04-04:
Try the "File->Example->01.Basics->Fade" example program. Put in your LED pin number and have a look. My library is calling the `analogWrite` function to do the PWM. If it is a core problem you will see.

One other option, are you realy sending just one frame for one universe? In case your program is continusly sending, pause it. There is a `Serial.print("D");` output prepared, remove the comment signs an see if "other" data is written to the output, which will result in flickering too.

Keep in mind, this is a simple example.

haaslukas, 2017-04-06:
Thanks for your fast reply. The basic PWM fade example is working fine, so I guess I'm just handling the continues DMX commands incorrectly. I'm using Q Light Controller+, which of course sends data all the time (isn't that the concept of DMX?). This also leads to lots of printed "D"s, when enabling `Serial.print("D");` and probably to the flickering because the ESP8266 is using `analogWrite` up to 44 times per second.

Do you have any idea how to handle multiple DMX commands or isn't the library intended to be used this way?

rstephan, 2017-04-06:
There is nothing wrong with sending data continuisly. The problem is the example ESP8266_Artnet_RGB_LED. One reason is the handling of the universe, this example didn't handle it. Take a look into ArtnetnodeWifiDebug, so you can see what data for what universe your program is transmitting to. The other reason can be, as you mentioned, calling `analogWrite` many times per second with the same value. This can result in flickering too, but i have build it up and can not confirm it.

Try the Debug-example and use this callback (your programm has to transmit for universe 1).
```
void onDmxFrame(uint16_t universe, uint16_t length, uint8_t sequence, uint8_t* data)
{
  if (universe != 1) {
    return;
  }
  analogWrite(pinR, data[0]);
  analogWrite(pinG, data[1]);
  analogWrite(pinB, data[2]);
}
```

haaslukas, 2017-04-06:
Yeah, the problem mostly exists between chair and keyboard. Of course I didn't check for the universe within the code and my DMX program was sending DMX values on four different universes. So the example code was writing four different analog values which lead to the flickering. Thanks a lot for the hint with the Debug script and the callback function you provided.

closed 2017-04-06 by haaslukas