rstephan/ArtnetnodeWifi #3: I have a problem using Artnet RGB LED

Protezhe, 2017-06-29:
Hi! Qlab detect node, but do not show universe of it. Leds working only in broadcast mode. And i can't create universe for esp node...

rstephan, 2017-06-30:
I have tested Qlab4 and the program is only transmitting in broadcast mode. It look like a problem in Qlab to me (or we both can't configure it right). If Qlab is not transmitting, no lights will change.
Sorry, it is not a problem with my ArtnetnodeWifi library. Ask the Qlab-manual for help.

Protezhe, 2017-07-01:
Thanks you for answer! :)
Its not only Qlab problem, because in "DMX-workshop" there is no UNIVERSE of NODE too :(
And in DMX workshop all parameters in RED and there is not "OUTPUT (universe)", but using other handmade Node parameters are green. And "OUTPUT" have Universe number
Qlab perfect working with nodes, if it's show universe

rstephan, 2017-07-03:
Good news!
I have updated the examples with a line of `artnetnode.setNumPorts(1);` The node has an output now. Previously it was not set (no output, no universe in Qlab).
After the fix you can see an universe of "0" in the Status-window.
Thanks for the hint.

Protezhe, 2017-07-03:
Perfect!! Thanks :)

closed 2017-07-04 by rstephan