rstephan/ArtnetnodeWifi #4: number of universe

Protezhe, 2017-07-30:
Hello, I was not careful last time.
Qlab show "0 universe" but I can't change it to other num... 
artnetnode.setStartingUniverse(2); - don't working, always "0 universe" show :(

rstephan, 2017-07-31:
I have fixed a problem in my library. Update the ArtnetnodeWifi library (1.0.4) and you have to add a few extra lines to you program.
```
  ...
  // set a starting universe if you wish, defaults to 0
  //artnetnode.setStartingUniverse(4);
  artnetnode.setNumPorts(2);
  artnetnode.enableDMXOutput(0);
  artnetnode.enableDMXOutput(1);
  artnetnode.begin();
  ...
```
Take a look into my examples, especially in `ArtnetnodeWifiDebug.ino`
QLab should show "Universes: 0, 1" now.

A Art-Net node can handle up to 4 outputs.

Protezhe, 2017-07-31:
Thanks! Now it's fully working :) 
Fantastic project ;)

closed 2017-08-23 by rstephan