rstephan/ArtnetnodeWifi #5: It is not issue

Protezhe, 2017-10-16:
Hi! I don't know how I can send you a message. Can artnetNode send artnet messages, like console? I want to create midi to artnet converter, but your ArtnetWifi is not working with my node...

rstephan, 2017-10-26:
Sorry for the long delay...
The `write()` function generates only the data-packet. For the most part the receiver needs to know that it can receive data from a master. They have to exchange poll-requests/responses.
But i will make some tests with the ArtNetominator, i only need more time ;-) ...

Protezhe, 2017-10-27:
Thank's) "ArnetnodeWifi" device on esp8266 is not controlled by "ArtnetWifi" transmit example.  "ArnetnodeWifi" - perfect library, maybe you can create Transmit example for it?

rstephan, 2017-10-28:
After some tests with Artnetdominator, i had no issues.
Have you set your correct destination in line 21 in ArtnetWifiTransmit, `const char host[] = "xxx"` ? The default value of 2.1.1.1 is propably wrong for your setup.
ArtnetWifiTransmit -> ArtnetnodeWifi sould be no problem as well.
Are you able to receive MIDI frames? If yes, then you can make a MIDI to ArtNet converter.
Adding a transmit-function to ArtnetnodeWifi, i would not bet on me ;-)
... but maybe if time goes by ....
Still not your answer, ask again!

Protezhe, 2017-10-28:
The default value of 2.1.1.1 is propably wrong for your setup.
How I can know the right value? 

I have no problems to receive midi messages, using midi library)

rstephan, 2017-10-28:
Look into Artnetdominator, it is the IP of your computer (the destination device).
E.g. http://www.lightjams.com/artnetominator/images/network.jpg

Protezhe, 2017-10-29:
Oh yes! It working))))) 
Maybe is the way to get IP of node automatically? Like it doing Qlab? Because using static IP on router less stablish. And change ip in code not operating) 
Or the way to use different IP together? If you have many phisical nodes?

rstephan, 2017-11-01:
The ArtNet Protocol is able to detect nodes automatically. For that are poll request/response frames there, as i mentioned earlier. ArtnetnodeWifi is able to response to poll requests, but can not generate a request. The target of development was to create a slave. The ability to transmit with ArtnetWifi (the other project) is only a simple extension, not a full featured node. If you need more functionality, so you have to write your own. Sorry.

Protezhe, 2017-11-01:
Thank's for answer! I will try to do it)

Protezhe, 2017-11-06:
Using ip 192.168.0.255 artnet send broadcast messages to all nodes! It is what I want)

closed 2017-11-10 by rstephan