rstephan/ArtnetnodeWifi #6: Artnetnode recive messages from different universes (Qlab + artnet transmit library)

Protezhe, 2017-12-07:
Hi! A few days ago I find new issue Artnetnode recive messages from different universes (Qlab + artnet transmit library)

In Artnetnode led on esp set universe 13, Qlab sent messages to it. But artnetnode recive zero udp from wifi transmit device on esp, set to 2 universe

wifi transmit send udp to broadcast adress 192.168.0.255

rstephan, 2017-12-07:
The broadcast address is the issue. The Art-Net specification says:

> ArtDmx packets must be unicast to subscribers of the specific universe contained in the
ArtDmx packet.

This is what the library enforces. If you are trying to send a ArtDmx packet with a broadcast address it will be dropped.

Protezhe, 2017-12-07:
Artnetominator identifi this messages correctly (right universe only light) and my artntet - dmx converter too. Maybe is the way to identifi universe in code for example:
if (universe 2){}

?

rstephan, 2017-12-08:
I have no doubt that the Art Net packet is correct. I will also believe that your Art Net DMX converter accept that message. (By the way, this is not what the specification says!)
But if this packet is send from a broadcast address i will drop it!
In case you send it from a unicast address i will accept it and after that you can easily to something like this:
```
if (universe == 2) { 
  DoSomething();
}
```

Protezhe, 2017-12-09:
writing: 'universe' was not declared in this scope

rstephan, 2017-12-09:
Take a look into the example `ArtnetnodeWifiDebug` :

```
void onDmxFrame(uint16_t universe, uint16_t length, uint8_t sequence, uint8_t* data)
{
  ...
  if (universe == 2) { 
    DoSomething();
  }
  ...
}
```

```
void setup()
{
  ...
  artnetnode.setArtDmxCallback(onDmxFrame);
  ...
}
```

Protezhe, 2017-12-09:
Thank's! Now it's workink perfect with broadcast messages! 

void onDmxFrame(uint16_t universe, uint16_t length, uint8_t sequence, uint8_t* data)
{

if (universe == 13) { 
    
   analogWrite(pin1, artnetnode.getDmxFrame()[511]);
      analogWrite(pin2, artnetnode.getDmxFrame()[511]);
  
  boolean tail = false;
  
  Serial.print("DMX: Univ: ");
  Serial.print(universe, DEC);
  Serial.print(", Seq: ");
  Serial.print(sequence, DEC);
  Serial.print(", Data (");
  Serial.print(length, DEC);
  Serial.print("): ");
  
  if (length > 16) {
    length = 16;
    tail = true;
  }
  // send out the buffer
  for (int i = 0; i < length; i++)
  {
    Serial.print(data[i], HEX);
    Serial.print(" ");
  }
  if (tail) {
    Serial.print("...");
  }
  Serial.println();
}
}

closed 2019-04-14 by rstephan