rstephan/ArtnetnodeWifi #8: How to obtain ArtNet message via Artpoll

iz1kbp, 2019-04-11:
Hi.Thank's in advance and sorry for my poor english.I've a question.How can I read long name and short name from ESP artnet node? Now I'm using your artnetwifi library to command my new esp2866 artnet interface.All works fine!!! In my old Arduino Artnet node(wired), using  broadcast (lan 2.x.x.x sub 255.255.255.0 ,an old artnetnode library) I send an art_poll command and I receive via wired lan all necessary data I've stored in Long and shor name (Ip address set in manual or auto mode, and more).Now with this new card, using Wifimanager,the problem is that I dont know his Ip address.I can discover his Ip only via  ROUTER (Wifi connected). Maybe a nice way If I must able to read Ip stored (long and shor name) in same manner that I use in arduino card, via OPCODE.How can I solve this problem?Thank's in advance  - Mario

iz1kbp, 2019-04-13:
Solved...Error on implementation in my scheck. Have a nice day!

rstephan, 2019-04-14:
See https://github.com/rstephan/ArtnetnodeWifi/issues/9

closed 2019-04-14 by rstephan