rstephan/ArtnetnodeWifi #9: Opcode trigger. How can I read and use?

iz1kbp, 2019-04-13:
Hi Stephan,
Sorry again..... I would kindly need to read the "TRIGGER" artnet opcode. I don't know how to modify your library, could you give me some directions? I need to change the status of an esp8266 digital output (Wemos d1 mini) by commanding via the Opcode trigger.Thanks any way and good continuation.
Mario

iz1kbp, 2019-04-14:
This my code put at the end of Loop.
If I use OpPoll or OpDmx my led blink and I have a serial readout....using OpTrigger nothing work.
------------------------------------------------
uint16_t code = artnetnode.read();
  //if (code) {
    if (code == OpTrigger) {
      Serial.print("trigger");
      digitalWrite(LED_B, LOW);
        delay(100);
      digitalWrite(LED_B, HIGH);
        delay(100);
      digitalWrite(LED_B, LOW); 
   // }
   }
-----------------------------

rstephan, 2019-04-14:
Hi Mario,
I have fixed a bug, regarding the return value of the `read()` function. Your code looks good to me. Can you try it please? I have nothing to test my modifications right now.
vy 73 Stephan

iz1kbp, 2019-04-14:
Hi Stephan..
Thank you for fast replay!!!
Yesssss...Good Job!!!
73 Mario

closed 2019-04-25 by rstephan