rstephan/ArtnetnodeWifi #10: Compiling error

Protezhe, 2019-05-14:
In new Arduino IDE  (Mac and windows) compile error

Arduino: 1.8.9 (Mac OS X), Плата:"Generic ESP8266 Module, 80 MHz, Flash, Disabled, ck, 26 MHz, 40MHz, DOUT (compatible), 512K (no SPIFFS), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

/Users/Alf/Documents/Arduino/libraries/ArtnetnodeWifi-master/src/ArtnetnodeWifi.cpp: In member function 'uint8_t ArtnetnodeWifi::begin()':
/Users/Alf/Documents/Arduino/libraries/ArtnetnodeWifi-master/src/ArtnetnodeWifi.cpp:55:40: error: ambiguous overload for 'operator~' (operand type is 'IPAddress')
   localBroadcast = IPAddress(localIP | ~localMask);
                                        ^
/Users/Alf/Documents/Arduino/libraries/ArtnetnodeWifi-master/src/ArtnetnodeWifi.cpp:55:40: note: candidates are:
/Users/Alf/Documents/Arduino/libraries/ArtnetnodeWifi-master/src/ArtnetnodeWifi.cpp:55:40: note: operator~(uint32_t {aka unsigned int}) <built-in>
/Users/Alf/Documents/Arduino/libraries/ArtnetnodeWifi-master/src/ArtnetnodeWifi.cpp:55:40: note: operator~(u32_t {aka long unsigned int}) <built-in>
/Users/Alf/Documents/Arduino/libraries/ArtnetnodeWifi-master/src/ArtnetnodeWifi.cpp:55:40: note: operator~(int) <built-in>

rstephan, 2019-05-16:
Thanks for pointing it out.
I can confirm the problem in Linux as well. It should be fixed by now.

Protezhe, 2019-05-18:
Thanks! Now compile good!!))

closed 2019-05-18 by Protezhe