rstephan/ArtnetnodeWifi #12: Esp32 performance unreliable

jaaaamIron, 2019-09-10:
Hi. I’ve noticed ESP32 performance to be very unreliable using this library compared to when I use it with ESP8266. Are there are ESP32 optimisations that are worth doing? I’m using NodeMCU 32S

rstephan, 2019-10-05:
I have tested my example `ArtnetnodeWifiDebug` all day long and can't see any crashes.
What is your setup? How do you handle the Art-Net data?

jaaaamIron, 2019-10-07:
Hi, thank you for your reply. ArtnetnodeWifiDebug also works for me.
I am running your example ESP8266_Artnet_RGB_Led, but modified to use the ledc code which is neccessary for ESP32 PWM. I am using the function artnetnode.getDmxFrame()[X], and I am retreiving 1-3 addresses. I get fairly regular freezes - probably one around every 5 or 6 minutes.