rstephan/ArtnetnodeWifi #15: ArtNet output

mrv96, 2020-02-08:
Hi, thank you for the great library.

Are you going to implement a write function like ArtnetWifi library in order to send out ArtNet?

rstephan, 2020-02-10:
The short answer would be no. But I have look up the Art-Net specification and a node should receive and/or transmit Art-Net DMX messages. So I guess, in the future it will be. The update will take some time, maybe some weeks or so.
In the meantime use my other library. Recently I have discovered some inconsistency in the transmit method, fixes are applied in a couple of days.

mrv96, 2020-02-10:
Ok, thank you very much! I tried your ArtnetWifi library and works great, but i have to wait for the update of this library because of this: https://github.com/rstephan/ArtnetWifi/issues/19

rstephan, 2020-03-21:
The time has come.
If you like, you can test the transmit function, example: `ArtnetnodeWifiTransmit.ino`

mrv96, 2020-03-22:
Hi, i tried your code and in principle it works, there are only two things that didn't convinced me.

1: The PollReply is not automatically managed by the library, is this wanted?
2: Usually in functions like setByte the user expects that the written values are saved in a dedicated memory, your function instead use the shared buffer.
Since there is a quite huge amount of RAM on ESP8266 IMHO a dedicated buffer would be better because saves a lot of instructions used to fill artnetPacket buffer at each loop(). Moreover usually ArtNet DMX programs need a dedicated buffer where store values either to send or to read, so that memory is usually allocated (either by the library or by the user).

rstephan, 2020-03-24:
To 1: The transmission of the ArtPollReply frames should be handled automatically. If not, than there is a bug. The content of that frame is up to the user, that is on purpose. But for convenient, it could be shown in the transmit-example how to do this (e.g. set the PortTypes field).

To 2: I have replicated the ArtnetWifi behavior, but you are right. A dedicated buffer for transmitter is a good idea.

Thanks for the advice!

mrv96, 2020-03-24:
As you can see in the pictures i'm not able to get a PollReply neither from "The ArtNetominator" nor QLC+.

[image]

[image]

rstephan, 2020-03-25:
The "missing" ArtPollReply is fixed.

Just a note. My `ArtnetnodeWifi` is just a library, not a ready to use product. The examples should demonstrate parts of the functionality, just to show how to use it. The Transmit-Example shows how to transmit. In this case transmit-only. Depending on how to look at the example, it may has missing features. To create your project, you have to combine and/or extend the examples to your needs.
So, from my point of view, it was not a bug.

The transmit buffer extension will come soon.

mrv96, 2020-03-25:
> 1: The PollReply is not automatically managed by the library, is this wanted?

Yes, I agree with you, in fact I asked. Maybe my question was a little bit mess, I should have used the word "example", not "library".

Thank you for your work!

closed 2022-10-04 by rstephan