rstephan/ArtnetnodeWifi #17: Prefered library?

solexious, 2021-02-20:
Hello!

I was looking for an esp32 artnet library and liked the way your's worked, then realised it's based off my old one for the launchpad! Was very excited to see others had been using my code, as it doesn't have the github fork link I didn't find this repo till now. I had tweaked it to run on the esp8266 a few of years back but never got around to publishing it.

Anyway, I'm looking to modify an artnet library to work with the ttgo ethernet poe esp32 board for wired ethernet artnet. I see you also have the https://github.com/rstephan/ArtnetWifi library. Is this or that library depreciated for the other? Is there one you would suggest using as a foundation over the other?

Cheers!

rstephan, 2021-02-20:
Hi
Yes, I have used our code. It was for a little project for a friend of mine. For the "new" ESP8266 modules back in the day. Because of the focus for WiFi and the lack of compatibility between wireless and wired network I simply renamed it. Some time later I decided to put it on github without much thought about forking and renaming. Maybe someone else will find it useful.
But to make the history of the code clear the note on the README-page was the way to go for me. Later, after the code became more popular, I questioned myself whether or not it was a good idea not to fork your project. But after that, it was too late. Sorry, I will do better next time. But now, i have a guilty conscience.

Some time ago I tested this code or the other one (ArtnetWifi) with a Olimex ESP32 wired ethernet module but I can't remember which one.

From my point of view, I am keeping both projects to serve different needs. This is the reason why both project still exists under my account.

* The ArtnetWifi is the simple one, just blinking lights as simply as possible. Most Arduino users don't need much, and some have very little programing experience , so less code is better. :-)
* ArtnetnodeWifi is the code-base for the more "sophisticated" one. To name just one feature, is it possible to discover the node (OpPoll).

I haven't done much to your code, just a few tweaks. Most of it is still yours, but as a foundation for a feature-rich node, I would use the ArtnetnodeWifi library.

solexious, 2021-02-20:
Oh, don't feel guilty at all! Github is a bit of a pain with forks, I hate that you can't use the search function with forked repos. (I still don't understand why) As you kept my historic copyright you are totally fine continuing it on in a new repo. In fact this is the first time I think I've had someone use something I coded as a base and improved on it, I was really happy to find someone thought it was good enough to work with! Yay for opensource, thank you :)

Cool to hear you had used it for that, when I used the library on the esp8266 I built these nodes, the esp8266 repo you link to was an early version of the firmware I used to make this: https://hackaday.io/project/11795/logs I should really update the repo.

Ahh, that makes sense, ArtnetWifi does seem to be a bit more popular on the platformio rankings. It might the ttgo board seems to do ethernet a bit oddly, will take a look into how best to merge it with ArtnetnodeWifi

Cheers!

closed 2021-02-21 by solexious