rstephan/ArtnetnodeWifi #18: Opcode unknow

jazzvivi2, 2022-03-04:
Hi, I love your project.  i have an unbranded chinese node that uses its own app to work.  I am trying to build another one with your library.  but the software does not detect it.  I clone all the data I found mac, name, manufacturer, etc but still not detected and looking at the traffic I found an unknown opcode that doesn't follow artnet rules, besides port changes.  I believe that the data of this opcode is what enables it to be recognized.  I can add that response somehow, create that opcode.  the photos with ip 0.1 are the software packages and those with ip 0.2 are those of my node
[image]

jazzvivi2, 2022-03-04:
[image]

jazzvivi2, 2022-03-04:
[image]

rstephan, 2022-03-05:
Something went wrong with uploading the images.
If your traffic was captured with wireshark, you can also copy/past **relevant** parts as Text/Hex-Dump.

jazzvivi2, 2022-03-05:
Thank you very much for answering, here I paste the information of the 4 packages. those from ip 2.0.0.1 are from the app. those from ip 2.0.0.2 are hardware. As I was saying, if you look at the ports they are intertwined, I think the app acts as a bridge. and there is the OpCode: Unknown (0x1005) which I think is what starts the app

Ip 2.0.0.1 software

User Datagram Protocol, Src Port: 6455, Dst Port: 6454
    Source Port: 6455
    Destination Port: 6454
    Length: 247
    Checksum: 0xba90 [unverified]
    [Checksum Status: Unverified]
    [Stream index: 32]
    [Timestamps]
    UDP payload (239 bytes)

Art-Net, Opcode: ArtPollReply (0x2100)
    Descriptor Header
        ID: Art-Net
        OpCode: ArtPollReply (0x2100)
    ArtPollReply packet
        IP Address: 2.0.0.1
        Port number: 6454
        Version Info: 0x0100
        NetSwitch: 0x00
        SubSwitch: 0x00
        Oem: Simon Newton: LibArtNet (0x0430)
        UBEA Version: 0
        Status: 0x00, Port Address Programming Authority: unknown, Indicator State: unknown
        ESTA Code: ESTA/PLASA (0x0000)
        Short Name: 4 Universes Input
        Long Name: 4 Universes Input Service
        Node Report: 0000 [0000] libartnet
        Port Info
        SwVideo: Displaying local data (0x00)
        SwMacro: 0x00
        SwRemote: 0x00
        spare: 000000
        Style: StNode (Art-Net to DMX device) (0x00)
        MAC: Broadcast (ff:ff:ff:ff:ff:ff)
        Bind IP Address: 0.0.0.0
        Bind Index: 0x00
        Status2: 0x00, Port-Address size: 8bit Port-Address
        filler: 0000000000000000000000000000000000000000000000000000

User Datagram Protocol, Src Port: 6455, Dst Port: 6454
    Source Port: 6455
    Destination Port: 6454
    Length: 22
    Checksum: 0x4a22 [unverified]
    [Checksum Status: Unverified]
    [Stream index: 32]
    [Timestamps]
    UDP payload (14 bytes)

Art-Net, Opcode: ArtPoll (0x2000)
    Descriptor Header
        ID: Art-Net
        OpCode: ArtPoll (0x2000)
        ProtVer: 14
    ArtPoll packet
        TalkToMe: 0x06, Send me ArtPollReply on change, Send diagnostics messages, Send diagnostics unicast: Broadcast
        Priority: DpAll (0)

Ip 2.0.0.2 hardware

User Datagram Protocol, Src Port: 6454, Dst Port: 6455
    Source Port: 6454
    Destination Port: 6455
    Length: 247
    Checksum: 0x4a6c [unverified]
    [Checksum Status: Unverified]
    [Stream index: 33]
    [Timestamps]
    UDP payload (239 bytes)

Art-Net, Opcode: ArtPollReply (0x2100)
    Descriptor Header
        ID: Art-Net
        OpCode: ArtPollReply (0x2100)
    ArtPollReply packet
        IP Address: 2.0.0.2
        Port number: 6454
        Version Info: 0x0100
        NetSwitch: 0x00
        SubSwitch: 0x00
        Oem: Simon Newton: LibArtNet (0x0430)
        UBEA Version: 0
        Status: 0xe0, Port Address Programming Authority: network, Indicator State: Normal Mode
        ESTA Code: ESTA/PLASA (0x0000)
        Short Name: AG_11277CE2F73C00
        Long Name: AG ArtNet4DMX512_Service
        Node Report: #1[2]RuningOK
        Port Info
            Number of Ports: 4
            Port Types
                Type of Port 1: Art-Net <-> DMX512 (0xc0)
                Type of Port 2: Art-Net <-> DMX512 (0xc0)
                Type of Port 3: Art-Net <-> DMX512 (0xc0)
                Type of Port 4: Art-Net <-> DMX512 (0xc0)
            Input Status
                Input status of Port 1: 0x00
                Input status of Port 2: 0x00
                Input status of Port 3: 0x00
                Input status of Port 4: 0x00
            Output Status
                Output status of Port 1: 0x80
                Output status of Port 2: 0x80
                Output status of Port 3: 0x80
                Output status of Port 4: 0x80
            Input Subswitch
                Input Subswitch of Port 1: 0x00
                [Universe of input port 1: 0]
                Input Subswitch of Port 2: 0x01
                [Universe of input port 2: 1]
                Input Subswitch of Port 3: 0x02
                [Universe of input port 3: 2]
                Input Subswitch of Port 4: 0x03
                [Universe of input port 4: 3]
            Output Subswitch
                Output Subswitch of Port 1: 0x00
                [Universe of output port 1: 0]
                Output Subswitch of Port 2: 0x01
                [Universe of output port 2: 1]
                Output Subswitch of Port 3: 0x02
                [Universe of output port 3: 2]
                Output Subswitch of Port 4: 0x03
                [Universe of output port 4: 3]
        SwVideo: Displaying local data (0x00)
        SwMacro: 0x00
        SwRemote: 0x00
        spare: 000000
        Style: StNode (Art-Net to DMX device) (0x00)
        MAC: 08:bb:e7:06:bf:da (08:bb:e7:06:bf:da)
        Bind IP Address: 2.0.0.2
        Bind Index: 0x01
        Status2: 0x08, Port-Address size: 15bit Port-Address
        filler: 0000000000000000000000000000000000000000000000000000

User Datagram Protocol, Src Port: 6455, Dst Port: 6455
    Source Port: 6455
    Destination Port: 6455
    Length: 76
    Checksum: 0xabfb [unverified]
    [Checksum Status: Unverified]
    [Stream index: 34]
    [Timestamps]
    UDP payload (68 bytes)

Art-Net, Opcode: Unknown (0x1005)
    Descriptor Header
        ID: Art-Net
        OpCode: Unknown (0x1005)
        ProtVer: 14
    Data: fa2e00c6cec9beb4bcf99fc7fa2e00c6cec9beb4bcf99fc7fa2e00c6cec9beb4bcf99fc7…

rstephan, 2022-03-05:
The first 3 packets looks like normal device discovery. The last packet is an unusual frame. Normal Art-Net traffic is on port 6454, but this one is on 6455 (Src and Dst) only.

Can you capture the response from the last packet, from the original hardware? And please include the full data part.

The partial data looks like a repetition of `fa2e00c6cec9beb4bcf99fc7`, many times.
I guess, this its the "magic" part for/from the App.

For now, my lib can only run on port 6454, as normal Art-Net devices do. But lets see what the real device replies, and maybe I can come up with a solution.

jazzvivi2, 2022-03-05:
i managed to make it work.  from what i could find out the software bridges receive on 6454 and sends node to 6455. that's why the node only works with this software.  copy these packets the standard and the unknown and forward them to port 6455. the app recognized a node.  As I thought, that package is the one who does the magic.  Now it remains to do it with the esp8266.  there is no response to that packet what it sends and receives is what I sent.  I am not in the office anymore but tomorrow I will send you the complete code

jazzvivi2, 2022-03-06:
full data

fa2e00c6cec9beb4bcf99fc7fa2e00c6cec9beb4bcf99fc7fa2e00c6cec9beb4bcf99fc7f74418f94025f74418f94025f74418f94025f3ff

rstephan, 2022-03-07:
I don't understand the bridging? There are only two parts (device and app), what must be bridged? 

Normally a device (node) responds with `ArtPollReply` to `ArtPoll` from the master (controller).

Your somewhat Art-Net node does presumably two things for `ArtPoll`:
- First: `ArtPollReply`
- Second `Unknown (0x1005)` with data `fa2e00c6...`

Am I right?

jazzvivi2, 2022-03-09:
if you're right. the app acts as a bridge between other apps and has an artnetominator-type display without this app turned on, the node does not work. I need to send Unknown (0x1005) with datafa2e00c6...

I don't know how to modify the library to send that data or if it would be easier to send the package in some way without modifying the library

rstephan, 2022-03-10:
Try this (**it's beta, not tested on hardware and I don't know the exact frame format!**):

I have also tweaked the library a bit to make it work, so check out the latest revision. 

For `loop()` do this
```
void loop()
{
  // we call the read function inside the loop
  uint16_t ret = artnetnode.read();
  if (ret == OpPoll) {
    SpecialSauce();
  }
}
```

And for the **special sauce**:
```
void SpecialSauce(void)
{
  WiFiUDP udp;
  IPAddress dst = artnetnode.getSenderIp();
  const uint16_t specialPort = ARTNET_PORT + 1;
  uint8_t artnetHeader[12];
  const uint8_t magicPacket[] = {
      0xfa, 0x2e, 0x00, 0xc6, 0xce, 0xc9, 0xbe, 0xb4, 0xbc, 0xf9, 0x9f, 0xc7,
      0xfa, 0x2e, 0x00, 0xc6, 0xce, 0xc9, 0xbe, 0xb4, 0xbc, 0xf9, 0x9f, 0xc7,
      0xfa, 0x2e, 0x00, 0xc6, 0xce, 0xc9, 0xbe, 0xb4, 0xbc, 0xf9, 0x9f, 0xc7,
      0xf7, 0x44, 0x18, 0xf9, 0x40, 0x25,
      0xf7, 0x44, 0x18, 0xf9, 0x40, 0x25,
      0xf7, 0x44, 0x18, 0xf9, 0x40, 0x25,
      0xf3, 0xff
    };

  memcpy(artnetHeader, ArtnetnodeWifi::artnetId, strlen(ArtnetnodeWifi::artnetId) + 1);
  uint16_t opcode = 0x1005;
  artnetHeader[8] = opcode;
  artnetHeader[9] = opcode >> 8;
  uint16_t version = 14;
  artnetHeader[10] = version >> 8;
  artnetHeader[11] = version;

  udp.beginPacket(dst, specialPort);
  udp.write(artnetHeader, sizeof(artnetHeader));
  udp.write(magicPacket, sizeof(magicPacket));
  udp.endPacket();
}
```