rstephan/ArtnetnodeWifi #19: DMX Signal Flickering

j0lsen, 2022-08-23:
I'm testing your SendDMX example on an Adafruit Huzzah ESP2866. The DMX output is coming from a Grove DMX512 adapter.

Everything is mostly working except the DMX signal is "flickering" - all channels momentarily drop to zero several times a second. Any thoughts on what may cause this?

rstephan, 2022-08-23:
Your setup looks like this:

```
PC --WiFi--> ESP8266 --TX-pin--> Grove Adapter --DMX--> Lamp1 --> Lamp2 --> ...
```

Right?

The ESP8266 is a 3.3V device.
The Grove DMX512 is a 5V device.
Is your power supply setup correct?

Some lights have a "black-out" function if they don't receive a frame in a certain interval.
Is your Art-Net source transmitting with a fixed packet length, so every light is always covert?
My SendDMX example is just repeating what's coming from the Art-Net.

j0lsen, 2022-08-24:
I think I've resolved the issue. I experimented with the length of the "break" signal in void onDmxFrame. Changing delayMicroseconds value to 300 gives me smooth signal. 

I was worried about the GroveDMX512 being underpowered. I tried a separate 5v supply with no change. So far it's functioning fine on 3.3v

Thanks for your response!

closed 2022-09-06 by rstephan