rstephan/ArtnetnodeWifi #20: Best practise to troubleshoot for signal ---> No Idea if I did set it up wrong or if my DMX broadcast is not working

vonfinck, 2023-02-17:
Hey, I am banging my head against this. 
I have used the example "send dmx ..." and I am using DMX Resolume Arena to broadcast a dmx signal to my fixtures. 
But I have no clue if I don't receive any signal on the esp32 or if I have setup the library wrong or if my dmx output is just wrong or not working. 

My setup:
AZ Delivery ESP32 + Resolume DMX Arena

ESP logs successfull connection to wifi and sends one "dmx data packet reveived" ping and then nothing more. 

How can I figure out where the stopper is? Or how would you test for positive signal in the wifi and signals received on the esp32? 

Thank you a lot

vonfinck, 2023-02-20:
Hey, BIG fat thanks already for the help! Setupup wise: Exactly like you put it. 

I am seeing now finally an output ;D 

DMX: Univ: 0, Seq: 0, Data (512): FF 37 34 0 0 0 0 0 0 0 0 0 0 0 0 0 ...
(changing of course constantly, )

Im now going to trouble shoot the RS485... maybe there is my flaw.

vonfinck, 2023-02-20:
I am getting further and further ;D 

Though the DMX decoder does not yet recognize the signal. 
Here is what I ca nscope from it:
[image]

Maybe the baud rate is not right? 
My esp32 supports only 3.3v output but has also a 5v pin. I assume for referenze voltage but not as power source / signal? 
I connected the RX485 as this:
RE and DE to the RS485's VCC pin (and that to the ESP32 5V Pin)
DE to ESP32 TXD pin

on the RS485 output the 
B to DMX D-
A to DMX D+
GND to common ground

rstephan, 2023-02-20:
That's good news so far.

For the wiring: Yes for the most part, the RS485 boards use a MAX485 or similar. 5V is the right voltage.
So:
- RE High (5V)
- DE High (5V)
- Vcc 5V

I guess the "**DE** to ESP" is a typo. "**DI** to ESP32 TX".
And yes, the ESP32 is a 3.3V device and the MAX485 is a 5V one. But in this direction, 3.3V out to an 5V input is for the most part okay.

For the A/B and D-/D+, there is no exact rule. If `A <-> D+` and `B <-> D-` is not working, try the other way around: `A <-> D-` and `B <-> D+`.

To the scope screen. Unfortunately, the top part of the screen is cut off.
A bit (the shortest time between a rising and falling edge) should be 4µs (1/250000 b/s).
On pin `DI/ESP TX` you should see the something like on the scope.
To measure the output, the RS485 signal, use channel 1 for pin A and channel 2 for B.
You should see two signals with, A is the inverse of B. If A goes up, B should go down and so on.
To "scope" digital signals, it is best to send only a few packets and also shot ones. No the ones with 512 bytes per frame.
Also use "single shot" to catch **A** frame and not many on top of each other.

And finally, check the address setting of the DMX receiver. According to the debug output you are only transmitting 3 (usefull) bytes (Addr. 1 to 3). Your other device must match.

vonfinck, 2023-02-24:
Hey i think somewhere I am loosing power or the wiring is wrong ;D. 
The Max485 also gets pretty hot. (untouchable hot)
I have setup more dmx fixtures, so that all available channels are beeing used. 
Also I took out a new Max485 to make sure that is not the issue. 

I have created a wiring diagram to check if I am doing it right. 
I am getting a signal at 3.2V at the TX Pin on the ESP32 but only 1.2 garbage at the A&B pin from the max485. 

Maybe I wired it wrong? 

[image]

[image]

vonfinck, 2023-02-24:
Alternative wiring I just tested ;D but no success...

[image]

rstephan, 2023-02-25:
The wiring looks all right.
3.2V at TX/DI pin is fine.
The A/B voltage looks a bit low, but this is a differential signal, so there is no need for a 5V signal.
But in case of so much heat, that sounds link a short circuit somewhere.
Maybe your MAX485 board/module has not survived your experiments, try a new one.
Disconnect all devices and check the A/B signal with your scope, as described earlier (with Art-Net/DMX traffic active).

Please check the polarity of the MAX485 (GND/VCC) and that you are connected to the DMX IN port.

Some DMX devices can be switch between master and slave. Check that all of your DMX devices run in slave mode.

Change the order of the DMX devices, to see if the behavior is the same.
With no DMX signal connected to a DMX device, it should be in "black out" mode. If a light is on, another device is transmitting data.

In a long chain of devices the last one must have the termination resistor active. Some need a dip-switch, others need an extra end-plug, better ones can do that automatically.

vonfinck, 2023-02-27:
Hey, I got the somewhat jumpy signal output to work. I think the Max485 did had a short, like you said.
It looks a bit like the signal is collapsing a few times a second. (I'll upload a video in a second) 
But I get a somewhat readible output with proper opposing signals. 
BUT so far I can not get the dmx decoder to accept it. 

I'll try to get a DMX tester to see if the decoder is at fault here. 

[image]

vonfinck, 2023-02-27:
Hey, I have tested it with an Arduino Uno and this library [https://github.com/PaulStoffregen/DmxSimple/blob/master/examples/SerialToDmx/SerialToDmx.ino](url)

I can successfully get the three channels 1;2;3 to light up ;). 
I think maybe it is an adress issue then?

rstephan, 2023-02-27:
That's good news. I wish you luck.

Just for curiosity, what was the fault? Bad wiring? Loose connections?

And one final question. The scope is showing the A/B complementary signal, but the time-base is 5µs/div. The pulses are 8µs wide -> 125kb/s. Am I right or did I miss something?

vonfinck, 2023-02-28:
Hey Stephan,

I think it was a mix of that.
I have in the last run replaced all jumper wires, new max485 module and
re-wired the dmx decoder connection.
The wiring diagram  i used now is the last one in the image. That worked
then finally.
Regarding the last question, I am going to measure that out more precisely
;D but i think you are right.
Let me check that today.