rstephan/ArtnetnodeWifi #21: more then 1 dmx output

tlundi60, 2023-03-08:
Hello and thanks for the great library
a question, I'm probably asking myself too stupid
the library supports multiple dmx outputs
how are the universes mapped to the exits
and can you define the output pins on the esp32?

rstephan, 2023-03-09:
Hi,
the library supports what you want.
It depends also on the hardware used (ESP8266, ESP32, RP2040, ...).

My example uses the UART of the controller to output the DMX signal. So the pins used are the UART pins of that specific controller. An ESP8266 has 2x TX and 3x TX on an ESP32 (if I am not mistaken).
There are other DMX libraries out there which uses a timer and interrupts to output signals on "any" pin.

How to map the universes also depends on the code in `onDmxFrame()`. In my example, all universes are output into the same DMX "bus". (Not very useful, but simple.)
I just provide examples, not ready-to-use code.

One can to code something like this:
```
void onDmxFrame(uint16_t universe, uint16_t length, uint8_t sequence, uint8_t* data) 
{
  HardwareSerial* out;
  switch (universe) {
  case 1: out = &Serial; break;
  case 2: out = &Serial1; break;
  default: return;
  }
  out->begin(56700);
  out->write((uint8_t)0);
  ...
}
```

tlundi60, 2023-03-10:
Hello and thanks for the quick reply and help

I will test it in the next days an tell the result
By the way it is correct esp32 has 3 x RX TX
Serial0: RX0 on GPIO3, TX0 on GPIO1
Serial1: RX1 on GPIO9, TX1 on GPIO10 (+CTS1 and RTS1)
Serial2: RX2 on GPIO16, TX2 on GPIO17 (+CTS2 and RTS2)

thank you again !

tlundi60, 2023-03-10:
Hello
just tried to activate the 2nd output
after changing the script, neither tx nor tx1 works anymore
do you have any idea why that is
here is the modified code
void onDmxFrame(uint16_t universe, uint16_t length, uint8_t sequence, uint8_t* data)
{

HardwareSerial* out;
  switch (universe) {
  case 1: out = &Serial; break;
  case 2: out = &Serial1; break;
  default: return;
  }

  // Send "break" as a "slow" zero.
  out->begin(56700);
 out->write((uint8_t)0);
  delayMicroseconds(300);
 out->begin(250000, SERIAL_8N2);
  
  out->write((uint8_t)0); // Start-Byte
  // send out the buffer
  for (int i = 0; i < length; i++)
  {
    out->write(data[i]);
  }
}

tlundi60, 2023-03-11:
Hello
i need help again
I have now solved it like this
the serial0 (RX/TX) work but serial2
does not work
I can't use Serial1 because the board keeps rebooting (GPIO connected to Flash)
I have already experimented with the settings in the void setup()
but no success
but the debug shows data on universes
here is my setup
void setup()
{//Starten der Schnittstellen
  Serial.begin(115200);
  Serial2.begin(115200, SERIAL_8N1,16,17);

  ConnectWifi();
  artnetnode.setName("ESP8266 - Art-Net");
  artnetnode.setNumPorts(2);
 // artnetnode.enableDMXOutput(0);
 //artnetnode.enableDMXOutput(1);
  artnetnode.begin();

  // this will be called for each packet received
  artnetnode.setArtDmxCallback(onDmxFrame);
}

tlundi60, 2023-03-11:
if i activate
artnetnode.enableDMXOutput(0);
artnetnode.enableDMXOutput(1);
then serial0 not work
what is the problem ?

tlundi60, 2023-03-11:
Hello
I have a result now
when I activate 3 universes in the artnet
"Art0;Art1;Art2"
then I also have an output on serial2
but only Art1 and Art2 are output Art0 not
What can be the reason for the art 0 not being issued

tlundi60, 2023-03-12:
Solved !
Hello again, thank you for the great library

everything works

my problem was that my universe 1 in magicq in the artnet is 0

thank you very much again !!

closed 2023-03-23 by tlundi60