rstephan/ArtnetnodeWifi #23: Fixed local broadcast address calculation and MAC address when in AP mode

alf45tar, 2023-06-09:
Hi @rstephan,
       I fixed your library when using EPS8266 in AP mode. Do you want me to submit a PR?

Thanks
alf45tar

rstephan, 2023-06-21:
Sure, why not.

rstephan, 2023-06-27:
Sorry for the delay. 

I don't like the fact, that there is a hard-coded value in the code.
Apart from that, it is looking good.

But you can get the CIDR with `softAPSubnetCIDR()` and compute the subnet mask form that.

alf45tar, 2023-06-29:
`softAPSubnetCIDR()` is not available for ESP8266.

alf45tar, 2023-06-29:
Hard coded netmask removed in AP mode using `wifi_get_ip_info()`.
PR ready to merge.

rstephan, 2023-07-10:
Hi,

and sadly, the function `wifi_get_ip_info()` is not available in ESP32. Or at least not in the Arduino board package 2.0.9. So this is also not a valid solution.

My library should work with every board or at least with as many as possible. 
Therefore, and due to the fact that every board (ESP8266, ESP32, RP2040, ...) hast a different API for the AP-Mode, I am going into a other direction. It is not a good solution to build exceptions for all controller into my library.

I am seeing this more like this:
```
uint8_t ArtnetnodeWifi::begin(String hostname); // "Normal", client mode
uint8_t ArtnetnodeWifi::begin(String hostname, IPAddress ip, IPAddress netmask, IPAddress gateway);
uint8_t ArtnetnodeWifi::begin(String hostname, IPAddress ip, IPAddress netmask, IPAddress gateway, byte* mac);
```

You have the information during setup for the `softAPConfig()` function, why not using this information for the `ArtnetnodeWifi::begin()` call?