rstephan/ArtnetnodeWifi #26: sendDMX() declared but not defined

danielkleinert, 2024-10-29:
I get the following compilation error:

```
undefined reference to `ArtnetnodeWifi::sendDMX()'
collect2: error: ld returned 1 exit status
```

Declared:
https://github.com/rstephan/ArtnetnodeWifi/blob/33bb66b532b4f880650622ee5208a656632d6c3a/src/ArtnetnodeWifi.h#L151

Used:
https://github.com/rstephan/ArtnetnodeWifi/blob/33bb66b532b4f880650622ee5208a656632d6c3a/src/ArtnetnodeWifi.cpp#L313

I am compiling in PlatformIO. Am I doing something wrong?

rstephan, 2024-10-30:
Hi @danielkleinert 

I have tested with: (It compiled fine.)
```
PlatformIO Core, version 6.1.16
PLATFORM: Espressif 32 (6.9.0) > Espressif ESP32 Dev Module
ArtnetnodeWifi @ 1.2.0
```

```
[platformio]
default_envs = esp

[env:esp]
platform = platformio/espressif32
framework = arduino
board = esp32dev
lib_deps = rstephan/ArtnetnodeWifi
```

Do you have more details for me?
What is you controller?

danielkleinert, 2024-10-30:
Hi @rstephan,

That's interesting. This is my setup:

```
platform = espressif32
board = lolin_c3_mini
board_build.filesystem = spiffs
framework = arduino
lib_deps = 
	tzapu/WiFiManager@^2.0.17
	rstephan/ArtnetWifi@^1.6.1
	bblanchon/ArduinoJson@6.21.4
```

After figuring out that my controller also allows Unicast, I switched to ArtnetWifi. So I sidestepped this issue. But it would be nice to have this fixed non the less. The board I have is an ESP32-C3 Super Mini.

```
❯ pio --version
PlatformIO Core, version 6.1.16
❯ pio pkg show espressif32

platformio/espressif32
Platform • 6.9.0 • Public • Published on Thu Sep 26 11:52:34 2024
```

Ah, yes, I use this PlatformIO fork: https://github.com/pioarduino/platform-espressif32

I have little to no experience in C++, but the definition of sendDMX() needs to be somewhere right? If it is missing in your codebase too, then maybe your compiler is more lenient or do you have an implementation of that member function that I could not find on my machine?

rstephan, 2024-10-31:
Hi @danielkleinert,

long story short, I have removed the `sendDMX()` method. Is was a leftover from a rework.
You are right, the method was not implemented, so it was up to the compiler to accept the "misbehavior".

For testing, I have used the "original" PlatformIO python package. And it also worked with your settings with the Risc-V controller.
I suspect the `pioarduino` uses a different compiler, or has a stricter configuration.
The Arduino-IDE also accepted the code. Hopefully the bug is fixed for you.